marmni/FreeCAD-PCB #4: Updating FreeCAD-PCB WB to be compatible with the upcoming 0.18 release (req. py3 and Qt5 compatibility)

luzpaz, 2018-09-03:
ref: https://github.com/FreeCAD/FreeCAD-addons/issues/93

luzpaz, 2019-12-05:
@marmni hi, just checking in since it looks like there is activity on this repo's 0.18 branch. Can you share what you're working on and what you intention for it is? Will it be compatible with FreeCAD v0.19 as well ?

marmni, 2019-12-09:
Hi,
at the moment I'm "fighting" with some old errors. Current version (branch 0.18) is fully usable and compatible with FC0.18, but at the moment it should be treated as a beta version. Some changes block backwards compatibility (day by day).

Major issues are currently resolved and I plan to release this version in a few weeks. After then I will check FC0.19.

luzpaz, 2019-12-09:
Fantastic!

marmni, 2019-12-15:
Hi,
at the moment these topics are prio and after solving them I will release final version:
- reimplementation names/values parameter for the parts
- 'Cut holes through all layers' - eliminate problems after reloading saved *.FCStd file
- reimplementation of the KiCAD importer/exporter.
- IDF importer - contour direction detection to avoid errors during importing a board

Wiredhome, 2019-12-20:
Hi Marmni, I have FC 0.18.4, and the latest FC-PCB with the updates from 3 days ago. The parts render nicely, but the board itself is not shown [Eagle]. I see this effect with the PCB\examples\example_1.brd as well as my own. My earlier combination FC 0.18 and the earlier FC-PCB for 0.18 did render the PCB, but with the known issues of the earlier combo.
Any thoughts for which file I could look at and compare to the older version?

marmni, 2019-12-22:
Hi,
I do not see any problems with rendering boards on FC 0.18.4. But to be sure, I sent all the files again. Please check now. Do you use win or linux?

Wiredhome, 2019-12-22:
**Fixed! And I'd say the best integration yet.**

[I'm on Win10.] I tried merging, but ended up with remove/reinstall FC 0.18.4 and then your latest integration, and it looks very good.

[image]

Minor Observations:
1. Some parts import all gray, even though opening their step file (or via File | Import) is as colorful as you might expect (same behavior as prior versions).
2) Some information printed in black, not white as expected.

marmni, 2019-12-23:
Nice to hear that :).

Ad. 2 - Have you set the 'Annotation' color in the settings? Edit->Preferences->PCB->Colors

[image]

Ad. 1 - There are few ways to solve this - because this is not a bug in PCB workbench.

Step 1. If you load *.step file to FC and you see colors -> you need to delete *.col file for mentioned model. Find part in the 'Assign models' window and use function -> 'Delete *. col files' (and reload board). Skip step 2 (if it helped).
Step 2. If you load *.step file to FC and you do not see colors -> you need to change STP export format: Edit->Preferences->Import-Export->STEP-> and select AP203. Load *.fcstd and export it as STEP with colors. Then follow step 1.

Wiredhome, 2019-12-24:
With your hints - I've learned a lot! More accurate presentation. Correct silk color (I had thought it would pull it from the .brd file).

[image]
Thanks again!

Wiredhome, 2019-12-24:
With your guidance - it just keeps getting and looking better.
[image]

Should I append other observations here, or open a new item?

marmni, 2019-12-26:
Hi,
I prefer one problem/proposal one topic - it is easier to manage them later.

marmni, 2020-01-02:
Done. Version for FC_0.18 and Py_3.6.

luzpaz, 2020-01-03:
@marmni it's not necessary to update the revision but I will do so anyway :)

closed 2022-09-09 by marmni