marmni/FreeCAD-PCB #5: missing module named PCBcheckFreeCADVersion

luemmel76, 2018-10-30:
Hi, I wanted to test the FreeCAD-PCB-FC_0.18 version, but the message comes immediately:
`Traceback (most recent call last):
  File "<string>", line 120, in InitApplications
  File "<string>", line 176, in <module>
ImportError: No module named PCBcheckFreeCADVersion
----------------------------------------------------------------------------------------------------
During initialization the error No module named PCBcheckFreeCADVersion occurred in F:\FreeCAD18\Mod\PCB\InitGui.py
Please look into the log file for further information
----------------------------------------------------------------------------------------------------
global name 'PCBcheckFreeCADVersion' is not defined
Traceback (most recent call last):
  File "<string>", line 115, in Initialize`

marmni, 2018-10-30:
Hi,
I know about this error, I just forgot to upload/update all files :). I will do it in a few minutes.

marmni, 2018-10-30:
Ok, done. Please replace file data/database.cfg by your file (do backup).

closed 2019-12-15 by marmni