marmni/FreeCAD-PCB #8: Board Edges not Creating

bricechart, 2019-04-22:
When using FreeCAD 0.18 build 16110 and FC_0.18 PY_3.6 the board edges are not importin from KiCAD.

gandy92, 2019-10-26:
I've had this problem too, using FreeCAD 0.18.3 and the latest FC_0.18. I've found and fixed a bug where importing the board outline stops with the first arc segment:
```
--- a/formats/eagle.py
+++ b/formats/eagle.py
@@ -150,7 +150,7 @@ class EaglePCB(mathFunctions):
             else:
                 [x3, y3] = self.arcMidPoint([i['x1'], i['y1']], [i['x2'], i['y2']], i['curve'])
                 arc = Part.Arc(FreeCAD.Vector(i['x2'], i['y2'], 0.0), FreeCAD.Vector(x3, y3, 0.0), FreeCAD.Vector(i['x1'], i['y1'], 0.0))
-                borderObject.addGeometry(self.Draft2Sketch(arc, borderObject))
+                borderObject.addGeometry(arc)
         ######
         for i in self.getCircles(dane, 20):
             borderObject.addGeometry(Part.Circle(FreeCAD.Vector(i['x'], i['y']), FreeCAD.Vector(0, 0, 1), i['r']))
```
Can you check if this also works for you?

marmni, 2019-10-30:
Hi,
I will install newest version of kicad and check what is going on.
@gandy92 I will check your solution.

marmni, 2019-11-04:
@gandy92 your solution + changing Part.Arc to Part.ArcOfCircle solved the problems. Thanks.

closed 2019-12-26 by marmni