marmni/FreeCAD-PCB #13: Database not converted

wilfridCEA, 2019-12-16:
Hello all,

Thanks for the FC0.18 branch unfortunately it seems that the database conversion failed (according to some post on sourceforge, I suppose there is one...) for me.
The created database.db file is empty. If I create a new component, all it's OK.

No specific error message in the report windows.

I tried "FreeCAD-PCB_database" without success.

Best regards,

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

marmni, 2019-12-16:
Hi,
for sure do not use "FreeCAD-PCB_database". I will check it.

marmni, 2019-12-16:
Hi,
problem should be solved, however you will loose all categories. Do you still have old FC? If yes - it is possible to implement them in new database format (without losing any connections with the models). If not - life is brutal :).

wilfridCEA, 2019-12-16:
Thanks a lot,

I tried directly with the 0.18, I did not lose the categories, not really sure, it seems to me that a few components are not anymore sorted in their category, need to compare to the database in 0.16.

The orientation of some components is modified, well it's not really critical.

Best regards, thanks a lot

Wilfrid

marmni, 2019-12-17:
[...]The orientation of some components is modified, well it's not really critical.[...]
So everything is ok - this version use different algorithm to calculate correct position of the parts (all parts will be shifted/rotated).

closed 2019-12-26 by marmni