marmni/FreeCAD-PCB #21: what is it..???

ffred, 2020-04-03:
Sorry, but I didn't see any description of what this module can do ?? what is it's purpose ?

hlovdal, 2020-04-12:
There is a file https://github.com/marmni/FreeCAD-PCB/blob/master/instruction.pdf that goes into depth about everything it can do, but I agree, the readme file severely lacks an overview of its purpose and capabilities.

ffred, 2020-04-13:
thanks..!
it's clear that without at least a basic description in the readme file, I'm sure that many will just go away and don't search for more (like I did the first time..).

marmni, 2020-04-13:
Hi,
I will add necessary info to the README.md file. About "instruction.pdf" - this is a very old file (but it contains almost all the necessary information), currently I'm preparing a new one -> https://github.com/marmni/FreeCAD-PCB/tree/testing/instructions (pre-alpha version).

luzpaz, 2021-08-01:
@marmni Please add README.md thanks!

goingalong, 2021-11-16:
The links to the instructions lead only to 404's (for me).

hlovdal, 2021-11-25:
@goingalong  The `testing` branch was deleted when pull request #52 was merged. The file(s) are now at https://github.com/marmni/FreeCAD-PCB/tree/master/instructions.

closed 2022-09-09 by marmni