marmni/FreeCAD-PCB #22: error importing PCB to FreeCad "Traceback (most recent call last)"

kokospalme, 2020-04-25:
When I try to open a pcb (no matter which one)
I get the following error:

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/PCBbrd.py", line 70, in open
    importBRD(filename, wersjaFormatu[0])
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/PCBbrd.py", line 230, in importBRD
    plytkaPCB = mainPCB(wersjaFormatu, filename)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/formats/PCBmainForms.py", line 94, in __init__
    self.wersjaFormatu = KiCadv4_PCB(filename, self)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/formats/kicad_v4.py", line 101, in __init__
    KiCadv3_PCB.__init__(self, filename, parent)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/formats/kicad_v3.py", line 95, in __init__
    self.dialogMAIN = dialogMAIN(self.fileName)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/formats/kicad_v3.py", line 57, in __init__
    self.layersNames = self.getLayersNames()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/PCB/formats/kicad_v3.py", line 81, in getLayersNames
    dane[106] = {"name": softLayers[self.databaseType][106]["name"], "color": softLayers[self.databaseType][106]["color"]}
<class 'KeyError'>: ('name',)`

And I have to clue what it could be. Do someone knows where this comes from?

I'm working on OSX with this versions:
- FreeCAD 0.18, Libs: 0.18R16146 (Git)
- FreeCad-PCB plugin: https://sourceforge.net/projects/eaglepcb2freecad/files/Version_for_FreeCAD_min_0.18_and_Python_min_2.7/
- Kicad version 5.1.4-0-10_14

marmni, 2020-04-25:
Hi,
my mistake (I forgot to change "name" to "description" :/). Fixed.
This is the most current version: https://github.com/marmni/FreeCAD-PCB

kokospalme, 2020-04-29:
thanks for the quick reply. It worked!

closed 2020-04-29 by kokospalme