marmni/FreeCAD-PCB #31: Feature Request - Dereference attributes

Wiredhome, 2020-07-05:
In an Eagle design, I can create global attributes 
[image]

Then on the PCB, I can place some text, perhaps in the top document layer, that references the attribute.
[image]

This is basically the same as the Name and Value on a component - other than where in the .brd structure it is.
FreeCAD-PCB does not dereference the attribute, so the silk screen on the PCB looks like this:
[image]

In the .brd XML, here's the global attributes definition:
`...
</libraries>
<attributes>
<attribute name="PRODUCT" value="Power Supply"/>
<attribute name="VERSION" value="0.1"/>
</attributes>
<variantdefs>
...`

And actually upward in the file is the reference to these attributes:
`...
<board>
<plain>
<wire x1="0" y1="0" x2="127" y2="0" width="0" layer="20"/>
<wire x1="127" y1="0" x2="127" y2="76" width="0" layer="20"/>
<wire x1="127" y1="76" x2="0" y2="76" width="0" layer="20"/>
<wire x1="0" y1="76" x2="0" y2="0" width="0" layer="20"/>
<hole x="5.6" y="5.6" drill="3.8"/>
<hole x="5.6" y="70.4" drill="3.8"/>
<hole x="121.4" y="70.4" drill="3.8"/>
<hole x="121.4" y="5.6" drill="3.8"/>
<text x="16.8" y="6.6" size="1.778" layer="21">&gt;PRODUCT</text>
<text x="16.7" y="3" size="1.778" layer="21">&gt;VERSION</text>
</plain>
<libraries>
...`

Is this an easy enhancement?

I am competent and SW, but never authored Python code. I was browsing into eagle.py, and in my version, starting at line 149:
`    def getPCB(self, borderObject):
        dane = self.getSection('plain')
        
        for i in self.getWires(dane, 20):
`

It looks like it could use an iterator like:
`        for i in self.getTexts(dane, 20):`

But from here I'm a bit out of my area of expertise, and I don't see a method for debugging other than FreeCAD.Console.PrintWarning, so I think it will take me some time to stumble up against a few dead-ends before I find a path forward.

marmni, 2020-07-05:
Hi,
added.

Wiredhome, 2020-07-05:
**Phenomenal!** and so much simpler than what was in my imagination. 

[image]

Thanks!

closed 2020-07-05 by Wiredhome