marmni/FreeCAD-PCB #37: Errors in FreeCAD V0.19

Panamasticks, 2020-11-11:
I had hoped to use this workbench / program to input *.BRD's into FreeCAD, view them and export as *.STEP files.   I had to upgrade FreeCAD to V0.19 before the Add-in Manager would show available workbenches.  I then installed the PCB workbench.  But the  first time I tried to open an Eagle *.BRD, I received the following error messages.  Is there anything I can do to fix this?  Please keep it simple as I have ZERO programming experience.  

14:33:41  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\PCBbrd.py", line 40, in <module>
    from PCBconf import *
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\PCBconf.py", line 31, in <module>
    from PCBfunctions import getFromSettings_Color_1, getFromSettings_Color
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\PCBfunctions.py", line 43, in <module>
    from PCBdataBase import dataBase
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\PCBdataBase.py", line 36, in <module>
    from sqlalchemy.ext.declarative import declarative_base
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\sqlalchemy\__init__.py", line 9, in <module>
    from .sql import (
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\sqlalchemy\sql\__init__.py", line 8, in <module>
    from .expression import (
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\sqlalchemy\sql\expression.py", line 33, in <module>
    from .visitors import Visitable
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\sqlalchemy\sql\visitors.py", line 28, in <module>
    from .. import util
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\sqlalchemy\util\__init__.py", line 8, in <module>
    from .compat import callable, cmp, reduce,  \
  File "C:\Users\jimc\AppData\Roaming\FreeCAD\Mod\pcb\sqlalchemy\util\compat.py", line 172, in <module>
    time_func = time.clock
<class 'AttributeError'>: module 'time' has no attribute 'clock'

marmni, 2020-11-12:
Hi,
delete PCB workbench and install one more time (eventually download from github). It should be ok now.

Panamasticks, 2020-11-12:
Thanks for your reply.   Delete or uninstall?  And if delete, where do the
files reside? I used the add-on manager to install the workbench, and it
didn't tell me where it installs.   Do I use the add-on mwnager to
reinstall?

Jim C.

> Hi,
> delete PCB workbench and install one more time (eventually download from
> github). It should be ok now.
>
>
> https://github.com/marmni/FreeCAD-PCB/issues/37#issuecomment-726169612

Jim Canale, Engineering Tech
Applied Concepts, Inc.
397 Route 281
Tully, NY  13159
work: (315)696-8876 x28
home: (607)749-3411

marmni, 2020-11-12:
Sorry my bad. Use option 'Install/Update selected' in the Addon manager (or uninstall and then install).

Panamasticks, 2020-11-13:
Thank you.  I'm away from my machine right now.  I will do as you suggest
when I am able to return.  Thank you for developing the software.

Jim C.

> Sorry my bad. Use option 'Install/Update selected' in the Addon manager
> (or uninstall and then install).
>
> https://github.com/marmni/FreeCAD-PCB/issues/37#issuecomment-726298786

Jim Canale, Engineering Tech
Applied Concepts, Inc.
397 Route 281
Tully, NY  13159
work: (315)696-8876 x28
home: (607)749-3411

closed 2022-09-09 by marmni