marmni/FreeCAD-PCB #44: SEGV loading Eagle BRD file

Zeb61, 2021-04-21:
OS: Linux OpenSUSE 15.2 64bit
FreeCAD version 0.19
FreeCAD-PCB installed from Add-on Manager

I try to load PCB file and I get SEGV signal. Why?
Before upgrade to 15.2 I had FreeCAD 0.16 perfectly working.

Thank you in advance for any help.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x394a0) [0x7f3adb8124a0]
#1  /usr/lib64/libexpat.so.1(XML_SetHashSalt+0x1b) [0x7f3aca4e846b]
#2  /usr/lib64/python3.6/lib-dynload/pyexpat.cpython-36m-x86_64-linux-gnu.so(+0x472f) [0x7f3a212ee72f]
#3  /usr/lib64/libpython3.6m.so.1.0(_PyCFunction_FastCallDict+0x90) [0x7f3adda7ab00]
#4  /usr/lib64/libpython3.6m.so.1.0(_PyCFunction_FastCallKeywords+0x50) [0x7f3adda7aec0]
#5  /usr/lib64/libpython3.6m.so.1.0(+0x12a582) [0x7f3addaec582]
#6  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x262e) [0x7f3addaef29e]
#7  /usr/lib64/libpython3.6m.so.1.0(+0x12a18f) [0x7f3addaec18f]
#8  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#9  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#10  /usr/lib64/libpython3.6m.so.1.0(+0x12a18f) [0x7f3addaec18f]
#11  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#12  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#13  /usr/lib64/libpython3.6m.so.1.0(+0x12a18f) [0x7f3addaec18f]
#14  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#15  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#16  /usr/lib64/libpython3.6m.so.1.0(+0x129fe6) [0x7f3addaebfe6]
#17  /usr/lib64/libpython3.6m.so.1.0(+0x12a25f) [0x7f3addaec25f]
#18  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#19  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#20  /usr/lib64/libpython3.6m.so.1.0(+0x129fe6) [0x7f3addaebfe6]
#21  /usr/lib64/libpython3.6m.so.1.0(+0x12a25f) [0x7f3addaec25f]
#22  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#23  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#24  /usr/lib64/libpython3.6m.so.1.0(+0x12a18f) [0x7f3addaec18f]
#25  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#26  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#27  /usr/lib64/libpython3.6m.so.1.0(+0x12a18f) [0x7f3addaec18f]
#28  /usr/lib64/libpython3.6m.so.1.0(+0x12a465) [0x7f3addaec465]
#29  /usr/lib64/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3d75) [0x7f3addaf09e5]
#30  /usr/lib64/libpython3.6m.so.1.0(+0x129fe6) [0x7f3addaebfe6]
#31  /usr/lib64/libpython3.6m.so.1.0(PyEval_EvalCode+0x23) [0x7f3addaec123]
#32  /usr/lib64/libpython3.6m.so.1.0(+0x153a70) [0x7f3addb15a70]
#33  /usr/lib64/libpython3.6m.so.1.0(PyRun_StringFlags+0x75) [0x7f3addb17555]
#34  0x7f3addf776d4 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib64/FreeCAD/lib/libFreeCADBase.so+0x64
#35  0x7f3adee698bc in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x8c
#36  0x7f3adee69a8f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0xff
#37  0x7f3aded985ec in Gui::Application::open(char const*, char const*) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x18c
#38  0x7f3adee7581c in StdCmdOpen::activated(int) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x55c
#39  0x7f3adee6d896 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x2f6
#40  0x7f3adc45fe35 in QMetaObject::activate(QObject*, int, int, void**) from /usr/lib64/libQt5Core.so.5+0x835
#41  0x7f3add2af822 in QAction::triggered(bool) from /usr/lib64/libQt5Widgets.so.5+0x42
#42  0x7f3add2b1e3c in QAction::activate(QAction::ActionEvent) from /usr/lib64/libQt5Widgets.so.5+0xcc
#43  /usr/lib64/libQt5Widgets.so.5(+0x30e03c) [0x7f3add42803c]
#44  /usr/lib64/libQt5Widgets.so.5(+0x31563b) [0x7f3add42f63b]
#45  0x7f3add4304b3 in QMenu::mouseReleaseEvent(QMouseEvent*) from /usr/lib64/libQt5Widgets.so.5+0x1b3
#46  0x7f3add2f6238 in QWidget::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x1c8
#47  0x7f3add43277b in QMenu::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0xbb
#48  0x7f3add2b5f2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x9c
#49  0x7f3add2be418 in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x11c8
#50  0x7f3adee36978 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x88
#51  0x7f3adc430328 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x118
#52  0x7f3add2bca0f in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib64/libQt5Widgets.so.5+0x1df
#53  /usr/lib64/libQt5Widgets.so.5(+0x1f75f4) [0x7f3add3115f4]
#54  /usr/lib64/libQt5Widgets.so.5(+0x1f9d6a) [0x7f3add313d6a]
#55  0x7f3add2b5f2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x9c
#56  0x7f3add2bd540 in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x2f0
#57  0x7f3adee36978 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x88
#58  0x7f3adc430328 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x118
#59  0x7f3adca37465 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib64/libQt5Gui.so.5+0x745
#60  0x7f3adca387b5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /usr/lib64/libQt5Gui.so.5+0x135
#61  0x7f3adca1223b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Gui.so.5+0xbb
#62  /usr/lib64/libQt5XcbQpa.so.5(+0x721da) [0x7f3acd3581da]
#63  /usr/lib64/libglib-2.0.so.0(g_main_context_dispatch+0x154) [0x7f3ad6848624]
#64  /usr/lib64/libglib-2.0.so.0(+0x529c0) [0x7f3ad68489c0]
#65  /usr/lib64/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f3ad6848a4c]
#66  0x7f3adc48d3af in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x5f
#67  0x7f3adc42e57a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x13a
#68  0x7f3adc437780 in QCoreApplication::exec() from /usr/lib64/libQt5Core.so.5+0x90
#69  0x7f3adeda3ef1 in Gui::Application::runApplication() from /usr/lib64/FreeCAD/lib/libFreeCADGui.so+0x15f1
#70  /usr/lib64/FreeCAD/bin/FreeCAD(+0x38a4) [0x55ebf9a618a4]
#71  /lib64/libc.so.6(__libc_start_main+0xea) [0x7f3adb7fd34a]
#72  /usr/lib64/FreeCAD/bin/FreeCAD(+0x45aa) [0x55ebf9a625aa]

marmni, 2021-05-16:
Hi,
unfortunately, this error does not tell me anything.

Zeb61, 2021-05-16:
What I can do to help in problem search? I think problem is around libexpat, but it's just an opinion.
Regards.

luzpaz, 2023-05-07:
Error should be put in code indentations because it's spamming other open github issues

closed 2023-05-14 by marmni

luzpaz, 2023-05-14:
@marmni recommend you edit this ticket so that the ticket links don't pollute any future tickets