marmni/FreeCAD-PCB #57: File open error

Jihadist, 2023-03-18:
Cannot open pcb file

> 19:33:38  Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "C:\Users\user\AppData\Roaming\FreeCAD\Mod\pcb\PCBbrd.py", line 63, in open
>     wersjaFormatu = wersjaFormatuF(filename)
>   File "C:\Users\user\AppData\Roaming\FreeCAD\Mod\pcb\PCBbrd.py", line 188, in wersjaFormatuF
>     projektBRD = builtins.open(filename, "r").read()
>   File "C:\Program Files\FreeCAD 0.19\bin\lib\encodings\cp1251.py", line 23, in decode
>     return codecs.charmap_decode(input,self.errors,decoding_table)[0]
> <class 'UnicodeDecodeError'>: 'charmap' codec can't decode byte 0x98 in position 3066: character maps to <undefined>

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

marmni, 2023-03-19:
Hi,
this *.pcb file is from gEDA PCB? I just checked and files from version 4.2.2. are importing correctly.

Jihadist, 2023-03-20:
Hi @marmni, I have p-cad 2006 pcb binary file... Looks like plugin doesn't know about this format. Because I convert it to current Altium format and I see 'unsupport error'.