marmni/FreeCAD-PCB #58: KiCAD file only imports traces

Issac-Archer, 2023-04-29:
I came across this addon trying to use a CNC to make a PCB design but I'm having trouble importing a PCB from KiCAD to get started.

[image]

The screenshot above shows the PCB in KiCAD's 3D viewer.  However when I import into FreeCAD all I get are the traces.  No pads, no holes, no board:

[image]

I tried various combinations of import options, first select all then picking just layers I want, but nothing seems to help.  I'm running KiCAD 6.0.9 and FreeCAD 0.20.2.  KiCAD PCB file attached below in case there's something wrong with that:
[Midfield Box V2.zip](https://github.com/marmni/FreeCAD-PCB/files/11360048/Midfield.Box.V2.zip)

marmni, 2023-05-01:
Hi,
I'm still working on it, but the basic elements should work/import.

You can replace 3 files (downloading everything is not necessary):
./PCBobjects.py
./formats/kicad_v3.py
./formats/kicad_v4.py

Issac-Archer, 2023-05-01:
That fixed it, thanks!

Some of the text on the silkscreen layer isn't translating onto the board right, I'm getting a lot of REF*** and misaligned labels, but since I just want to cut and drill I'm not too worried about it.